steff393/wbec #125: wbec generating different mqtt devices under fhem

marc107, 2024-03-17:
Hello everyone,

i have been using wbec in combination with mqtt and fhem for some time now and I noticed that several (over 100) mqtt devices have been created for wbec in fhem - see the following screenshot. Of course, the device with the highest consecutive number is always used, but deleting the more than 100 devices is not really fun.
Does anyone have an idea why new devices are always being created in autocreate? Apparently wbec always reports to the mqtt broker with a different name!

Greetings & thanks

Marc

[image]

abuesen, 2024-03-18:
Hi Marc,

I'm using the Mosquito MQTT-Brooker. That's fine, there is one topic `wbec` and one `OpenWB` published by the wbec. No different names here. 

Does your wbec use a fixed IP in your network? 

Regards, Andre. 

[image]

marc107, 2024-03-20:
Hi Andre,

thanks for your reply!
Yes, IP of wbec is fixed.
I have to explore the mqtt tomorrow, currently i don't know the newest topic generated @fhem. There are a lot... ;-)

marc107, 2024-03-21:
I did some searching and found out that the mqtt-server included in fhem creates a new device with every reconnect if no clientid is defined in the mqtt device (here the wbec).

Excerpt from the fhem help:
Most MQTT-capable devices contain options for assigning a unique ClientID (see the example of the zigbee2mqtt service above). If no ClientID is assigned, some clients use new IDs for each connection. It is recommended to make use of these setting options if possible.

Unfortunately I could not find the option, is it not included in wbec or does it have a different name?

steff393, 2024-03-23:
Hello, sorry, but I don't have further information about the MQTT client id.

abuesen, 2024-03-24:
> Hello, sorry, but I don't have further information about the MQTT client id.

Die Arduino-Library unterstützt eine ClientID: https://github.com/knolleary/pubsubclient
Dokumentiert in der API: https://pubsubclient.knolleary.net/api#connect

steff393, 2024-05-01:
Ich lasse den Issue mal offen und baue künftig eine parametrierbare Client-ID ein: cfgMqttClientId
Wenn der Parameter 0 ist, dann wäre das Verhalten wie bisher (=random(255)). Ansonsten der eingestellte Wert.

closed 2025-01-05 by steff393