steff393/wbec #126: Ladestrom auf 0A in der Weboberfläche im Modus „PV Laden aus“

ivolein, 2024-03-23:
Hallo,

ist es Absicht , dass man im Modus „PV laden aus“ den Ladestrom nicht auf 0A setzten kann und das Laden somit beenden kann?

wenn ich den Regler auf 0 setzte, wird er autom. Wieder auf den vorherigen Wert gesetzt.
Beim wechsel von „pv“ aus „aus“ ist der Regler bei 0A und muss hier auf > 6A verschoben werden.

über das Web-Json kann man 0A einstellen.

steff393, 2024-03-24:
Hallo,
im Modus „PV laden aus“ sollte man den Strom ganz normal über den Schieberegler verstellen können.
Evtl. ist die Zeitsteuerung (oder eine andere Funktion) im Hintergrund noch parallel aktiv.

ivolein, 2024-03-25:
Hallo,
ich habe dieses Verhalten noch etwas weiter untersucht und festgestellt, dass dies vom Browser abhängt.

Mit dem Chrome auf dem PC kann ich den Ladestrom auf "0A" stellen. Auf dem Iphone mit dem Safari und Chrome kann ich nur zw. 6A und 16A regeln - 0A wird nicht übernommen. Der Slider zeigt einen Wert von 0A an - bleibt aber intern auf den letzen Wert der >= 6A war.

VG, Michael

oldlok, 2024-05-08:
Hallo,

habe folgendes bei der Verwendung von PV-Überschussladen festgestellt:
- Sind für den Bezug des Überschusses über MQTT in der config eine cfgMqttIp und die anderen Parameter eingetragen und der Broker sendet keine Daten, so lässt sich über die weboberfläche manuell kein Ladestrom einstellen. Es werden keine '0'-Werte angezeigt, sodern '-'. Auch bei dekativiertem Überschussladen (cgfPvActive = 0) bleibt dies so, solange eine IP für den Topic eingetragen ist.
- bei Verwendung des Überschussladens (egal ob mit oder ohne min-Strom) wird bei Vorhandensein eines Überschusses der Mindestladestrom getriggert, danach erhöht sich der Ladestrom in 3 oder 4 Schritten bis auf 16A (max), ohne dass der Überschuss dies rechnerisch hergeben würde.

Grüße
KS

steff393, 2026-09-22:
Mit Version 1.5.6 kommt demnächst ein komplett überarbeitetes Web Interface. Dort sollte das Verhalten behoben sein.

closed 2026-09-22 by steff393