steff393/wbec #127: wbec keine Verbindung möglich

csh-de, 2024-03-31:
keine Verbindung möglich, es kommt immer

Die Website ist nicht erreichbar 
192.168.178.25 hat die Verbindung abgelehnt.

warum passiert das? Warum lehnt er Verbindungen ab?
Ping geht aber das web Interface geht nicht auf.
Ping hat 4ms und Signal ist bei 90%, wenn ich wlan reset mache und neu einrichte, aber komme nicht mehr auf die Web interface.

steff393, 2024-03-31:
s. Issue #99

closed 2024-03-31 by steff393

abuesen, 2024-03-31:
> keine Verbindung möglich, es kommt immer
> 
> Die Website ist nicht erreichbar 192.168.178.25 hat die Verbindung abgelehnt.
> 
> warum passiert das? Warum lehnt er Verbindungen ab? 

Wie schon geschrieben wahrscheinlich issue 99, ein zu viel für den ESP. Es tritt hauptsächlich mit Mobilgeräten, auf. Vom PC sollte funktionieren.