steff393/wbec #129: cfgEnergyOffset | keine Veränderung bei den Werten

Heino1988, 2024-04-11:
Hallo 
bei mir hängt sich die Wallbox sporadisch auf, sodass sie blau blinkt und erst durch eine komplette Kappung der Stromversorgung wieder in den regulären Modus kommt.

Dabei ist mir am Energiezähler aufgefallen, dass dieser um eine gewisse, teilweise sehr hohe, Strommenge zurückgesetzt wird - siehe Datenbank-Grafik.
[image]

Da ich damit auch eine Stromaufstellung für meine Firma mache, würde ich gerne diese Strommenge über den cfgEnergyOffset wieder hinzuaddieren.
Weder ein positiver, noch ein negativer Wert in den Configs ändert bei mir etwas an der Energiemenge.

Ist das nur bei mir so bzw. kann ich das Thema mit #128 zusammenlegen?

Danke und Grüße
Heino

abuesen, 2024-04-11:
Hallo Heino,

aus dem https://github.com/steff393/wbec/issues/128 habe ich auch gelernt, dass der Zähler nur beim Übergang in _Standby_ gespeichert wird. Sehr doof.  Ich habe auch einen eigenen Zähler davor. Deine Frage kann ich nicht beantworten.

Es wäre aber vielleicht eine Idee, für Grafana die einzelnen Ladevorgänge anstatt der Gesamtmenge zu speichern. Also möglichst das Delta pro Zeiteinheit. Damit kann man alle späteren Fragestellungen auch wieder herstellen. Auch mit dem Offset würde dem Zähler der HDEC im absoluten Wert nicht annähernd vertrauen. Die Deltas dürften statistisch deutlich zuverlässigere Werte erzeugen.

Wenn es für eine Abrechnungen verwendet wird muss eh ein geeichter MIT-Zähler daher. Kostet etwa 60,- plus Einbau.

LG, A.

csh-de, 2024-04-14:
ja das hängt mit dem Standby zusammen..
die Info hatte ich vom Entwickler bekommen, da ich es mir erst nicht erklären konnte.. 

also lieber mit Standby arbeiten dann sind die Werte nicht weg...

Heino1988, 2024-04-15:
> also lieber mit Standby arbeiten dann sind die Werte nicht weg...

Wo und wie kann ich das aktiv ansteuern?

csh-de, 2024-04-15:
[image]

im config der weboberfläche

csh-de, 2024-04-15:
ich hätte ja gerne Standby an von 8-14uhr und den Rest ohne Standby, weil mit ist die Erreichbarkeit bei mir eingeschränkt... aber dennoch funktioniert es und das ist mir wichtiger als das die Verbrauchs-Zähler nicht gespeichert würden...

steff393, 2024-04-15:
Hi, mit der aktuellen SW x.5.2 ist der Parameter cfgStandby nicht mehr relevant. Stattdessen geht http://wbec.local/json?standby=0 (aktiv) oder = 4 (inaktiv). Meine Empfehlung bleibt weiterhin der Wert 4.
Der Energiezähler ist lt. Heidelberg-Anleitung nicht für Abrechnungszwecke geeignet. 
Wenn die Box im Standby ist, ist keine Kommunikation möglich, also auch keine Änderung desselben. Da das Thema schon komplex genug ist (z.B. für Lastmanagement und EnWG) wird auch keine zeitliche Abhängigkeit umgesetzt, sondern man muss sich für das eine odere andere entscheiden und mit den Konsequenzen leben.

closed 2024-04-15 by steff393