steff393/wbec #130: Laden nicht freigegeben

Gomercity, 2024-04-17:
Hallo,
ich habe heute den wbecPro an 2 Heidelberg Energy Control installiert. Vorher waren die beiden Boxen über das interne Lastmanagemeng verbunden.
Alle Jumper nach Anleitung eingestellt. Beide Wallboxen werden auch erkannt. Wenn ich ein Fahrzeug einstecke (egal ob Box 1 oder 2 oder beide) erkennt der wbecPro, dass ein Ladekabel eingesteckt wurde (Verbunden: Ja, ohne Ladeanf.), bei Laden erlaubt steht "nein". Was mache ich falsch? PV-Überschussladen ist deaktiviert, cfgCntWb":2, alles jeweils mit Reset bestätigt. Wieso gibt der wbecPro der Wallbox nicht das Signal, dass sie laden darf?
Vielen Dank für die Antwort.

Timo

Gomercity, 2024-04-17:
der Schieberegler steht übrigens auf 0A, lässt sich zwar verschieben, springt dann aber immer wieder zurück auf 0A

steff393, 2024-04-17:
Hi,
nutzt du auch den Parameter cfgTotalCurrMax?
Falls ja dann erteilt das Lastmanagement der Wallbox erst dann einen höheren Strom als 0 zu, wenn das Auto eine Ladeanforderung sendet (chgStat = 6 oder 7). 
Scheinbar senden manche Autos aber erst eine Ladeanforderung, wenn die Wallbox einen Strom > 0 erlaubt --> klassischer Deadlock...
In SW x.5.2 gibt es einen zusätzlichen Parameter cfgLmChargeState. Wenn man den auf 4 stellt, dann erlaubt das Lastmanagement bei einem chgStat von 4-7 (also ab "Fzg. angesteckt") einen höheren Strom.

closed 2024-04-18 by Gomercity