steff393/wbec #131: JSON Sonnenbatterie

Gomercity, 2024-04-18:
Lieber Stefan,
ich möchte nun den wbecPro dazu bringen, PV-Überschussladen durchzuführen.
Ich habe eine Sonnenbatterie Eco 8.0, über die Abfrage "http://192.168.178.56:80/api/v1/status"
erhalte ich folgende Werte:
"Apparent_output": 3009,
  "BackupBuffer": "0",
  "BatteryCharging": false,
  "BatteryDischarging": true,
  "Consumption_Avg": 7310,
  "Consumption_W": 7324,
  "Fac": 50.0130004882813,
  "FlowConsumptionBattery": true,
  "FlowConsumptionGrid": true,
  "FlowConsumptionProduction": true,
  "FlowGridBattery": false,
  "FlowProductionBattery": false,
  "FlowProductionGrid": false,
  **"GridFeedIn_W": -2570,**
  "IsSystemInstalled": 1,
  "OperatingMode": "2",
  "Pac_total_W": 2997,
  "Production_W": 1758,
  "RSOC": 27,
  "RemainingCapacity_Wh": 1685,
  "Sac1": 1006,
  "Sac2": 998,
  "Sac3": 1004,
  "SystemStatus": "OnGrid",
  "Timestamp": "2024-04-18 15:50:54",
  "USOC": 23,
  "Uac": 237,
  "Ubat": 49,
  "dischargeNotAllowed": false,
  "generator_autostart": false,
  "NVM_REINIT_STATUS": 0

Mich interessiert ja lediglich die Einspeisung in das Netz, also der Wert: "GridFeedIn_W": -2570
Im wbecPro habe ich folgende Werte hinterlegt:

cfgPvActive":1,"
cfgPvHttpIp":"192.168.178.56","
cfgPvHttpPath":"/api/v1/status","
cfgPvHttpJson":"GridFeedIn_W","
cfgPvHttpPort":80,"

eigentlich müsste mir doch im Menü im Feld "Bezug/Einspeisung" ein Wert angezeigt werden, was mache ich falsch?
Vielen Dank für eine weitere Hilfe.

Timo

Gomercity, 2024-04-20:
Hallo,
ich habe im Suchstring cfgPvHttpJson bei "GridFeedIn_W": den Doppelpunkt vergessen, Fehler korrigiert, nun funktioniert das PV-Überschussladen auch einwandfrei mit einer Sonnenbatterie. Vielleicht für andere Interessant die Vorgehensweise.
Sonnenbatterie über 192.168.x.x direkt ansteuern. Anmelden als "USER", Passwort "sonnenUser3552"
Dann unter "Software-Integration" den JSON API freischalten (in diesem Fall reicht Read API). Der Pfad und die Abfragemöglichkeiten werden angezeigt: http://192.168.xxx.xx:80/api/v2/latestdata
Bei V2 benötigt man einen Token, dieser wird ebenfalls angezeigt (curl --header 'Auth-Token: ef0d......'), da ich nicht wusste, wie man einen Token im wbecPro korrekt einträgt habe ich die Abfrage über API V1 durchgeführt.
cfgPvActive 1
cfgPvHttpIp 192.168.xx.xx (eigene IP eintragen)
cfgPvHttpPath /api/v1/status
cfgPvHttpJson "GridFeedIn_W": (wichtig, hier mit Anführungszeichen und Doppelpunkt)
cfgPvHttpPort 8080

Vielleicht werde ich für diesen Post belächelt, aber es gibt bestimmt auch viele da draußen, die keine Ahnung von JSON haben und mit dieser Anleitung die Sonnenbatterie mit dem wbecPro verbinden können.

steff393, 2024-04-20:
Hallo,
vielen Dank für die ausführliche Anleitung, bin mir auch sicher, dass sie anderen hilft.
Ich hab sie gleich im Wiki hinzugefügt: https://github.com/steff393/wbec/wiki/PV-%C3%9Cberschussladen#anbindung-an-sonnenbatterie-ab-v149

Gomercity, 2024-04-21:
Hallo Stefan,
zu erwähnen ist noch, dass cfgPvInvert auf 1 gesetzt werden muss. Dann stimmen die Vorzeichen für Bezug(+) / Einsp.(-).

Als Beobachtung des heutigen Tages. Der Wert "GridFeedIn_W": gibt ja lediglich die Netzeinspeisung in Watt an. D.h. bei genügend Sonne startet der wbecPro die Wallbox. Wenn sich nun eine Wolke am Himmel auftut und sich die PV-Leistung reduziert, dann gibt die Sonnenbatterie mit seinen maximalen 3KW Strom ins Hausnetz. Erst wenn (PV-Produktion) + (maximale Einspeisung der Sonnenbatterie) unter die Ladeleistung der Wallbox fällt, dann regelt wbecPro die Ladeleistung herunter. Bedeutet in der Praxis, dass permanent die Sonnenbatterie entladen wird, um das KFZ zu laden. Das möchte ich natürlich nicht, da damit ja unnötig viele Ladezyklen der Sonnenbatterie produziert werden.
Habe dann nicht mehr die Einspeisung ins Netz ("GridFeedIn_W":) als Bezugswert genommen, sondern die PV-Produktion ("Production_W": ). Damit entlädt sich nicht mehr die Batterie, außer die Wärmepumpe springt an, dann wird diese wieder über Batteriestrom der Sonnenbatterie gespeist - möchte ich auch nicht. 
Ideal wäre also folgende Formel:

Maximale Einspeisung ins Netz abzüglich Entladung der Sonnenbatterie
("GridFeedIn_W":  ) - ("Pac_total_W":  )

Dann würde immer nur soviel Strom an die Wallbox abgegeben werden, dass sich die Sonnenbatterie tagsüber nicht entlädt. Beide Werte werden über das API der Sonnenbatterie abgefragt (siehe Post oben). Im Config-Menü des wbecPro kann ich nur einen Bezugswert eintragen. 
Kannst Du mir helfen, wie ich diese beiden Werte nach o.a. Formel im wbecPro eintragen kann? Dann wäre die Steuerung mit der Sonnenbatterie perfekt!

Vielen Dank im Voraus und noch einen schönen Restsonntag.
Timo (Gomercity)

abuesen, 2024-04-21:
Hi Timo,

Das gleiche Problem hatte ich letztes Jahr auch. Sinkt der PV-Ertrag, springt die Batterie in die Bresche und jede Überschussregelung bekommt das nicht mit.

Deshalb deaktiviere ich inzwischen die Batterie, wenn das Auto geladen wird. Dafür scheint es bei Deiner Anlage eine Steuerung zu geben: `"dischargeNotAllowed": false`.

Das wbec kann aber rechnen noch das "Entladen der Batterie abschalten" setzen, soweit ich weiß. Das müsstest du also beides extern lösen, wenn Steff da nicht doch schon was für hat. 
@steff393 so ein http-Aufruf bei Begin und Ende des Ladens wäre eine simple Erweiterung für Hausbatterien, die per API steuerbar sind.

@Gomercity vermutlich wirst du aber demnächst weitere Bedürfnisse haben. Im Winter  zum Beispiel lade ich schon bei 2kW Überschuss und hole den Rest aus den Netz, weil alternativ sonst alles aus dem Netz käme und die 2kW ungenutzt verpuffen (oder wie man die Einspeisevergütung nennen mag). Deshalb nutze ich inzwischen evcc, aber ausschließlich für die Laderegelung. (der Rest hat mir nicht zugesagt). Es gibt auch andere gute Regler. Der wbec bedient den modbus und regelt für den Notfall aber einfacher, falls das andere System mal ausfällt.

Was du möchtest (rechnen) könnte man mit NodeRed, Homeassistant oder ioBrooker lösen. Je nachdem, wer die Sonnenbatterie gut unterstützt. Den wbec dann über http-aufruf oder mqtt einbinden. 

LG, Andre. 
LG, Andre.

steff393, 2024-04-21:
Hi,
ich glaube du brauchst nach der Umstellung auf die PV-Leistung "Production_W" auch noch die Änderung von "cfgCalcMode":1, s. [hier](https://github.com/steff393/wbec/wiki/PV-%C3%9Cberschussladen#berechnung).

Es gibt auch den "cfgCalcMode":2, um die Batterie zu berücksichtigen, z.B. bei Huawei, SAX und Goodwe. 
Allerdings muss wbec dafür die Batterieladeleistung kennen. Ein Weg dafür wäre [hier](https://github.com/steff393/wbec/wiki/PV-%C3%9Cberschussladen#anbindung-eigener-systeme-via-http) beschrieben. Der ist aber in deiner Situation vermutlich nicht machbar.

Aktuell ist nur die Abfrage eines HTTP-Wertes vorgesehen. Lass den Issue einfach mal offen. Ich überleg mir, wie ich es einbauen kann. Entweder 2 HTTP-Aufrufe oder 2 Suchstrings ...

Gomercity, 2024-04-21:
Hallo Stefan,
das wäre großartig.
Danke!
Timo

Gomercity, 2024-04-21:
Hallo Andre,
vielen Dank für Deine Ausführung. Batterieentladung manuell stoppen habe ich auch überlegt, jedoch wäre mir die Formel ("GridFeedIn_W": ) - ("Pac_total_W": ) am liebsten, dann braucht man keine Batterie wieder einschalten etc., da sind mir zuviele Fehlermöglichkeiten.
Gruß
Timo

hbpv, 2024-05-12:
Hi

Als weiteren kleinen Batterie Workaround für alle mit HomeAssistant: ich stelle per Automatisierung sobald sich der ChargingState/Modus von wbec ändert, auf meinem Goodwe Wechselrichter den DOD auf 2% (theoretisch ginge wohl auch 0%), somit stoppe ich zumindest das „Einspringen“ der Batterie.

Nachteil: Laufen in der Zeit weitere energieintensive Verbrauche, ziehen die natürlich auch aus dem Netz.

Never change a Running System hat mich bis jetzt davon abgehalten den BatteryValue von wbec zu nutzen :-D

steff393, 2025-01-05:
Im nächsten Release von wbecPro wird es den cfgPvHttpJsonBatt dafür geben.

closed 2025-01-05 by steff393