steff393/wbec #135: Not able to get value for "Bezug(+)/ Einsp.(-")

Timmy2807, 2024-05-12:
Ich habe meine Energy Control von Heidelberg erfolgreich verbindne können und kann auch über die wbec.local Seite die Steuerung grundsätzlich vornehmen, allerdings bekomme ich den Wert "Bezug(+)/ Einsp.(-") nicht (mehr angezeigt). Ich vermute eine fehlerhafte Einstellung in der Config, finde allerdings meinen Fehler nicht. Die Einbindung in OpenWB hatte ich schon versucht, wollte aber zuerst das Grundsätzliche aufsetzen, bevor ich mich an weitere Einbindungen versuche..

`{"cfgApPass":"wbec1234","cfgCntWb":1,"cfgMqttIp":"\"\"","cfgMqttWattTopic":"","cfgPvActive":1,"cfgPvHttpPath":"","cfgInverterIp":"192.168.178.xxx","cfgInverterType":4,"cfgInverterPort":502,"cfgChargeLog":1}`

Ich nutze eine Heidelberg Energy Control in Verbindung mit einem Huawei Wechselrichter und dem SDongle.

abuesen, 2024-05-13:
Wie erfährt der wbec denn den Wert für Einspeisung / Bezug in Deiner Umgebung? Hat es schon mal funktioniert? 

In der config sieht man, dass mqtt gar nicht konfiguriert wurde. (Damit könnte es sich die Werte mittels `cfgMqttWattTopic` holen.) Alternativ ruft ein externes Programm das passende Kommando am wbec auf, siehe Doku. 

Falls du openWB bereits hast versuche die Anbindung darüber. Das geht auch über mqtt. 
Huawei und SDongel kenne ich nicht.

Timmy2807, 2024-05-14:
Danke @abuesen für deine Rückmeldung! 

Tatsächlich hatte ich es schon einmal sporadisch zum Laufen bekommen. Aber seit ich OpenWB per MQTT anbinden wollte, funktioniert es nicht mehr. Hatte dann versucht, alles wieder auf die damaligen Einstellungen zu bekommen, aber seitdem bekomme ich den Wert gar nixht mehr angezeigt. In der Zwischenzeit konnte ich OpenWB über MQTT grundsätzlich einbinden und mache mich mal an deinen vorgeschlagenen Weg über `cfgMqttWattTopic `.

Danke für deine Hilfe!

steff393, 2024-06-24:
Hi,
ich bin mir nicht ganz sicher, woran es liegt, aber man sollte die Übermittlung von Bezug/Einspeisung auf einen Weg festlegen:
- MQTT (z.B. auch openWB): cfgMqttIp und cfgMqttLp sind nötig
- HTTP-Aufrufe (z.B. Shelly): cfgPvHttpIp, cfgPvHttpPath und ggf. cfgPvHttpJson sind nötig
- Modbus-TCP (z.B. Huawei): cfgInverterIp und cfgInverterType sind nötig

Oben sind Ansätze aller 3 Methoden in der cfg.json, evtl. hat das was vermischt.
Aber es scheint sich ja mittlerweile gelöst zu haben. Sorry für die späte Rückmeldung

closed 2024-06-24 by steff393