steff393/wbec #136: Short-Term Charging After Plugging In / Kurzzeitiges Laden nach dem Einstecken

raddis666, 2024-05-27:
ich habe einen Verbesserungsvorschlag für die Wallbox-Software, der möglicherweise auch anderen Nutzern helfen könnte. Ich besitze einen Smart Fortwo ED und lade mein Elektroauto hauptsächlich über meine PV-Anlage. Dabei bin ich auf folgendes Problem gestoßen:

Problem:
Wenn mein Auto eingesteckt wird und die Wallbox aufgrund fehlenden Solarstroms das Laden nicht sofort freigibt, geht das Auto nach einer Weile in den Schlafmodus. Das führt dazu, dass das Auto nicht automatisch zu laden beginnt, sobald wieder Solarstrom verfügbar ist. Dies ist besonders problematisch, da das Auto erst aufwacht und mit dem Laden beginnt, wenn es nach dem Einstecken kurz Strom bekommen hat.

Vorschlag:
Es wäre hilfreich, wenn die Wallbox-Software eine Funktion implementieren könnte, die nach dem Einstecken des Autos für eine kurze Zeit (z.B. 10 Minute) das Laden ermöglicht, um das Auto am Einschlafen zu hindern. Danach kann die Wallbox wie gewohnt den Ladevorgang basierend auf der Verfügbarkeit von Solarstrom steuern.

Technische Details:
Auto: Smart Fortwo ED
Ladesituation: Laden über PV-Anlage, keine externe Stromversorgung
Erforderliche Funktion: Kurzzeitiges Laden (ca. 1 Minute) nach dem Einstecken, um den Schlafmodus des Autos zu verhindern

Ich denke, dass diese Funktionalität über ein kurzes Setzen des Stromwerts nach dem Einstecken realisiert werden könnte. Wenn die Wallbox für eine kurze Zeit (z.B. 10 Minute) nach dem Einstecken des Autos einen minimalen Ladestrom (z.B. 6-7 A) bereitstellt, bleibt das Auto wach und kann später, wenn Solarstrom verfügbar ist, direkt mit dem Laden beginnen.

Vielleicht lässt sich diese Funktion in einer zukünftigen Version der Wallbox-Software berücksichtigen. Vielen Dank im Voraus für eure Unterstützung und großartige Arbeit!

Beste Grüße,
Raddis

I have a suggestion for the Wallbox software that might be beneficial to other users as well. I own a Smart Fortwo ED and primarily charge my electric car using my PV system. I have encountered the following issue:

Problem:
When my car is plugged in and the Wallbox does not immediately allow charging due to the lack of solar power, the car goes into sleep mode after a while. This results in the car not automatically starting to charge when solar power becomes available again. This is particularly problematic because the car will only wake up and start charging if it receives a short burst of power after being plugged in.

Suggestion:
It would be helpful if the Wallbox software could implement a function that allows the car to charge for a short period (e.g., 10 minute) after being plugged in to prevent it from going into sleep mode. After this short charging period, the Wallbox can then manage the charging process based on the availability of solar power as usual.

Technical Details:
Car: Smart Fortwo ED
Charging Situation: Charging via PV system, no external power supply
Required Function: Short-term charging (about 10 minute) after plugging in to prevent the car from entering sleep mode
I believe this functionality could be implemented by setting the current value to a minimal charging level (e.g., 6-7 A) for a short period (e.g., 10 minute) after the car is plugged in. This would keep the car awake and ready to start charging as soon as solar power is available.

I hope this feature can be considered for a future version of the Wallbox software. Thank you in advance for your support and excellent work!

Best regards,
Raddis

steff393, 2024-06-24:
Hallo Raddis,
evtl. lässt sich das auch aktuell schon über cfgPvMinTime realisieren. Bin aber grad nicht sicher, ob das beim ersten Einschalten (wenn kein Überschuss vorhanden ist) auch hilft. Ansonsten bliebe noch der Modus "Min+PV", das wäre aus meiner Sicht das Mittel der Wahl bei einschlafendem Auto.
Ansonsten einfach mal den Issue offen lassen, vielleicht fällt mir noch was ein.

steff393, 2026-09-22:
Hat zwar etwas gedauert, aber so eine Funktion kommt ab der Version 1.5.6 (cfgPvWakeTime). Damit kann man ein periodisches kurzes Laden (cfgPvMinTime) einstellen.

closed 2026-09-22 by steff393