steff393/wbec #137: Anbindung Sungrow Inverter ModBus TP

N4zroth, 2024-05-30:
Hi,
ich würde gerne einen PR stellen in dem die Anbindung an einen Sungrow SH10RT-V112 ermöglicht wird. Ich lese sowieso schon diverse Daten des Inverters aus um Grafana/Influx zu befüllen. Leider verstehe ich die Variablen in inverter.cpp eher weniger und der Inverter unterstützt leider kein Sunspec.
Außer dass pwrMet irgendwie das Ergebnis der Berechnung ist, blicke ich nicht wirklich durch. Gibt's da irgendwo eine Doku oder einen Hinweis zu? Danke!

steff393, 2024-05-30:
Hi,
die Unterstützung für Sungrow Wechselrichter ist bereits für die Pro-Varianten eingebaut:
https://steff393.github.io/wbec-site/

Das kommt im nächsten Release (1.5.3 bzw. 2.5.3), allerdings auf die wbecPro, wbecLan und wbecPremium. Eine Testversion kann ich vorab gern zur Verfügung stellen.

closed 2024-05-30 by steff393

N4zroth, 2024-05-30:
Hi,
oh super, hab's gerade auch eingebaut. Danke für die schnelle Info, dann bestelle ich demnächst eine Pro oder Premium :)

HowardH7, 2024-09-11:
Hallo,
ist eine Einbindung von Sungrow in den "normalen" wbec auch geplant?

steff393, 2024-09-11:
Leider nein, da unterscheidet sich aktuell der Code ziemlich stark. Sungrow nutzt anders als die anderen sog. Input-Register.

HowardH7, 2024-09-11:
> Leider nein, da unterscheidet sich aktuell der Code ziemlich stark. Sungrow nutzt anders als die anderen sog. Input-Register.

Danke für die schnelle Antwort.