steff393/wbec #139: WBEC does not connect back to Wifi Network after power outage

Chaosflori25, 2024-06-16:
My Wifi Network takes a bit, but it happend a few times in the last year, and I'm tired of running arround replugging the ESP after the Outage. Any Ideas how I could fix that?

Best Regards

steff393, 2024-06-24:
Hi,
there is the following parameter:
uint16_t cfgWifiConnectTimeout;       // Timeout in seconds to connect to Wifi before change to AP-Mode

Default is 10s

closed 2024-06-24 by steff393

andreas5232, 2025-06-18:
I've just stumbled upon the same issue. After power outage my wifi access point in the garden needs more than 10s to initialize and wbec is "stuck" in AP mode. Took me some time to figure it out... 

Maybe a more conservative default value of 120s is a better default value for cfgWifiConnectTimeout?