steff393/wbec #140: Bug in feature: Load log: Instead of 20, up to 50 entries can now be read if required (log.js, line 46 -> len=50)

sam-harder, 2024-06-17:
If the log list has been expanded to 50 entries, new loads will not be displayed. When the list is restricted to 20 lines again, the most recent loads appear on top.

steff393, 2024-06-24:
Hi, 
can you please send me your chargelog-0.txt file via wbec393@gmail.com?
Then I'll try to reproduce.

sam-harder, 2024-06-24:
Hallo,Sorry, beim Versuch diese Datei herunterzuladen habe ich sie scheinbar zerschossen. Zeigt nun völligen Unsinn an. Wie bekomme ich diese Datei gefahrlos heruntergeladen?Kann ich diese Datei einfach löschen?Viele Grüße Hans-Joachim Harder

steff393, 2024-06-24:
Hi, das ist ganz normal, dass die Datei sehr seltsam aussieht, wenn man sie im Editor oder unter wbec.local/edit anschaut.
Kein Problem.

steff393, 2024-06-24:
Rechtsklick --> Download

sam-harder, 2024-06-24:
Hallo,bin auf dem IPad unterwegs und finde leider keine Option den Download zu starten.Aber bei meinen unbedarften Versuchen das hinzubekommen, habe ich wohl den Inhalt der Datei verändert, denn wenn ich das Log aufrufe, sind darin unsinnige Werte zu sehen. (Hänge ein Bild an)Gruß Sam

closed 2024-09-14 by steff393