steff393/wbec #143: Auto lädt nicht mehr automatisch

amichael77, 2024-07-19:
Hallo,
bis vor ein paar Monaten wurde der Ladevorgang automatisch gestartet nachdem der Stecker eingesteckt war.
Jetzt ist es so, dass man auf der wbec-Seite explizit den Schieberegler nach rechts ziehen muss oder Min+PV auswählen muss, erst dann beginnt der Ladevorgang.

Müsste ich raten, würde ich behaupten, das ist so seit ich mal mit der Zeitsteuerung rumgespielt habe. Habe dort wieder alles deaktiviert lt. GUI; es ist auch nur eine Vermutung.

Habe einen BMW i3, HeidelbergEnergyControl und wbecPro 1.5.2. technisch hat sich meines Wissens da nichts geändert.

Wie bekomme ich den Ladevorgang wieder automatisch gestartet ohne jedes Mal klicken zu müssen?

danke im Voraus, Gruß Andi

steff393, 2024-07-21:
Hi Andi,
was kommt denn als Antwort auf http://wbec.local/time ?
Und auf http://wbec.local/json ?

amichael77, 2024-07-21:
Hallo Stefan,

folgendes kommt da: 

http://wbec.local/time:
`<html><body>
<div id="content"><div class="tabs tabs-tall "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="JSON durchsuchen" value=""></div></div><div class="panelContent" id="json-scrolling-panel" tabindex="0">
  |  
-- | --
mOn | 0
mOff | 0
cOn | 0
cOff | 0
flag | 0
eNow | 0
eDel | 0
eSta | 0
oOn | 0
oOff | 0
msm | 1165
unx | "19:25:20"

</div></div></div></div>
</body>
</html>`

http://wbec.local/json:
`<html><body>
<div id="content"><div class="tabs tabs-tall "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="JSON durchsuchen" value=""></div></div><div class="panelContent" id="json-scrolling-panel" tabindex="0">
  |  
-- | --
wbec |  
version | "v1.5.2"
bldDate | "Apr  6 2024 10:58:39"
timeNow | "19:24:01"
enwg14a | 0
enwgErr | 0
box |  
0 |  
busId | 1
version | "0"
chgStat | 0
currL1 | 0
currL2 | 0
currL3 | 0
pcbTemp | 0
voltL1 | 0
voltL2 | 0
voltL3 | 0
extLock | 0
power | 0
energyP | 0
energyI | 0
energyC | 0
currMax | 0
currMin | 0
logStr | ""
wdTmOut | 0
standby | 0
remLock | 0
currLim | 0
currFs | 0
lmReq | 255
lmLim | 0
resCode | "e4"
failCnt | 250
modbus |  
state |  
lastTm | 87608068
millis | 87614817
rfid |  
enabled | false
release | false
lastId | ""
pv |  
mode | 3
watt | 115
wifi |  
mac | "E8:9F:6D:0D:45:38"
rssi | -88
signal | 24
channel | 11

</div></div></div></div>
</body>
</html>`

steff393, 2024-07-22:
Hallo Andi,
ich denke, dass die Verbindung zur Wallbox unterbrochen ist. Darauf deuten die vielen 0en und resCode=e4. Bitte nochmal die Verkabelung/Schalter prüfen.

closed 2024-08-04 by steff393