steff393/wbec #144: Lastmanagement cfgTotalCurrMax Auto startet Ladevorgang nicht

DrPeppi, 2024-08-03:
Hallo in die Runde,

eventuell kann mir jemand helfen.

Sobald ich bei CurrMax was eingebe wird der Ladevorgang nicht gestartet. Ich habe die aktuelle Software installiert und ChargeState ist auf 4 geändert.

Eigentlich sollte das Problem mit der letzten Update behoben sein, bei mir jedenfalls startet der Ladevorgang nicht.

Kann mir jemand von euch helfen?

vielen Dank

LG Steffen

DrPeppi, 2024-08-03:
{
    "wbec": {
        "version": "v1.5.2",
        "bldDate": "Apr  6 2024 11:03:27",
        "timeNow": "20:19:08",
        "enwg14a": 0,
        "enwgErr": 0
    },
    "box": [
        {
            "busId": 1,
            "version": "108",
            "chgStat": 2,
            "currL1": 0,
            "currL2": 0,
            "currL3": 0,
            "pcbTemp": 295,
            "voltL1": 238,
            "voltL2": 241,
            "voltL3": 239,
            "extLock": 1,
            "power": 0,
            "energyP": 0,
            "energyI": 1314.027,
            "energyC": 0,
            "currMax": 16,
            "currMin": 6,
            "logStr": "00.779.2909/01 018590 2141 HD_EnergyControl",
            "wdTmOut": 60000,
            "standby": 0,
            "remLock": 0,
            "currLim": 0,
            "currFs": 0,
            "lmReq": 255,
            "lmLim": 0,
            "resCode": "0",
            "failCnt": 0
        },
        {
            "busId": 2,
            "version": "108",
            "chgStat": 4,
            "currL1": 0,
            "currL2": 0,
            "currL3": 0,
            "pcbTemp": 299,
            "voltL1": 242,
            "voltL2": 241,
            "voltL3": 241,
            "extLock": 1,
            "power": 0,
            "energyP": 0,
            "energyI": 1437.75,
            "energyC": 0,
            "currMax": 16,
            "currMin": 6,
            "logStr": "00.779.2909/02 042792 2221 HD_EnergyControl",
            "wdTmOut": 60000,
            "standby": 0,
            "remLock": 1,
            "currLim": 160,
            "currFs": 0,
            "lmReq": 160,
            "lmLim": 0,
            "resCode": "0",
            "failCnt": 0
        }
    ],
    "modbus": {
        "state": {
            "lastTm": 14633183,
            "millis": 14644197
        }
    },
    "rfid": {
        "enabled": false,
        "release": false,
        "lastId": ""
    },
    "pv": {
        "mode": 1,
        "watt": 0
    },
    "wifi": {
        "mac": "D4:D4:DA:5C:8F:90",
        "rssi": -61,
        "signal": 78,
        "channel": 6
    }
}

steff393, 2024-08-04:
Hi,
vermutlich liegt das Problem hier:
https://github.com/steff393/wbec/blob/5b6b69ea509baf483dd16b5354b7529e80c736ec/src/loadManager.cpp#L76
content[0][15] hat als Einheit nicht 0,1A wie die anderen Stromwerte, sondern 1A. Damit müsste hier dahinter nochmal ein " * 10".
Aktuell begrenzt die Funktion den Strom auf 1,6A (also effektiv 0A) statt auf 16A.

Außerdem kam (nur) in der Version 1.5.2 ein kleiner Copy&Paste-Fehler rein, der dazu führt, dass der Strom der 2. Box auf 0 gesetzt wird, wenn nur an dieser ein Fahrzeug angeschlossen ist (also dein oben beschriebener Fall).

Bugfix kommt in der nächsten Version.

DrPeppi, 2024-08-04:
Hi, 

vielen Dank für die schnelle Hilfe.
Momentan bin ich unterwegs und kann es erst in 1 1/2 Wochen testen.

LG Steffen

DrPeppi, 2024-08-04:
Dir vielen Dank für die schnelle Rückmeldung.Momentan bin ich unterwegs und kann es erst in 1 1/2 Wochen testen.Ja ich hatte dort 160 für 16 Ampere eingetragen.Verstehe ich das richtig das ich durch den bug 1600 hätte eintragen müssen das der Ladevorgang startet?LG Steffen

https://github.com/steff393/wbec/blob/5b6b69ea509baf483dd16b5354b7529e80c736ec/src/loadManager.cpp#L76

steff393, 2024-08-04:
Nein, den Bug kann nur eine neue Software beheben. Leider...

closed 2024-09-14 by steff393