steff393/wbec #145: Can't Build latest release

cothin, 2024-09-06:
Hello and thank you for sharing this software with us. 

I just cloned the latest release and tried to build it. 
Since AsyncElegantOTA was not automatically installed via Plattformio, i added the githup repo in the dependencies.

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nodemcuv2]
platform = espressif8266@4.0.1
board = nodemcuv2
framework = arduino
board_build.filesystem = littlefs
build_flags =
    -DWEBSOCKETS_SERVER_CLIENT_MAX=2      ; default was 5, but every additional client needs ca. 208 byte on heap
		-D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED
		-DWBEC_VERSION_MAJOR=0
lib_deps = 
	emelianov/modbus-esp8266@^4.1.0
	me-no-dev/ESP Async WebServer@^1.2.3
	bblanchon/ArduinoJson@^6.19.4
	knolleary/PubSubClient @ ^2.8
	arduino-libraries/NTPClient @ ^3.2.1
	miguelbalboa/MFRC522 @ ^1.4.10
	highno/RTCVars @ ^0.1.1
	links2004/WebSockets @ ^2.3.7
	;ayushsharma82/AsyncElegantOTA @ ^2.2.7
	https://github.com/ayushsharma82/AsyncElegantOTA.git
	thomasfredericks/Bounce2 @ ^2.70
monitor_speed = 115200
;lib_ldf_mode = deep+     ; was needed for ESP Async WebServer to avoid missing ESP8266Wifi.h (which is in fact not missing)
upload_speed = 921600
```

Nevertheless i am not able to compile. There could be something regarding the webserver.
Is this a known issue or am I missing something?
I really would appreciate your advice on this.

Here is the full compilation output:

```
 *  Executing task: C:\Users\cothi\.platformio\penv\Scripts\platformio.exe run 

Processing nodemcuv2 (platform: espressif8266@4.0.1; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Installing git+https://github.com/ayushsharma82/AsyncElegantOTA.git
git version 2.46.0.windows.1
Cloning into 'C:\Users\cothi\.platformio\.cache\tmp\pkg-installing-t1_wm01a'...
remote: Enumerating objects: 24, done.
remote: Counting objects: 100% (24/24), done.
remote: Compressing objects: 100% (19/19), done.
remote: Total 24 (delta 0), reused 14 (delta 0), pack-reused 0 (from 0)
Receiving objects: 100% (24/24), 406.75 KiB | 3.91 MiB/s, done.
Library Manager: AsyncElegantOTA@2.2.9+sha.5f48b2e has been installed!
Library Manager: Installing thomasfredericks/Bounce2 @ ^2.70
Unpacking  [####################################]  100%
Library Manager: Bounce2@2.72.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (4.0.1) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- modbus-esp8266 @ 4.1.0
|-- ESP Async WebServer @ 1.2.4
|-- ArduinoJson @ 6.21.5
|-- PubSubClient @ 2.8.0
|-- NTPClient @ 3.2.1
|-- MFRC522 @ 1.4.11
|-- RTCVars @ 0.1.1
|-- WebSockets @ 2.6.0
|-- AsyncElegantOTA @ 2.2.9+sha.5f48b2e
|-- Bounce2 @ 2.72.0
|-- LittleFS @ 0.1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoOTA @ 1.0
|-- WiFiManager-asyncwebserver
|-- EspSoftwareSerial @ 6.12.7
|-- ESP8266WiFi @ 1.0
|-- ESP8266HTTPClient @ 1.2
|-- SPI @ 1.0
|-- ESPAsyncTCP @ 1.2.2
Building in release mode
Compiling .pio\build\nodemcuv2\src\button.cpp.o
Compiling .pio\build\nodemcuv2\src\globalConfig.cpp.o
Compiling .pio\build\nodemcuv2\src\goEmulator.cpp.o
Compiling .pio\build\nodemcuv2\src\inverter.cpp.o
Compiling .pio\build\nodemcuv2\src\loadManager.cpp.o
Compiling .pio\build\nodemcuv2\src\logger.cpp.o
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Compiling .pio\build\nodemcuv2\src\mbComm.cpp.o
Compiling .pio\build\nodemcuv2\src\mqtt.cpp.o
Compiling .pio\build\nodemcuv2\src\phaseCtrl.cpp.o
Compiling .pio\build\nodemcuv2\src\powerfox.cpp.o
Compiling .pio\build\nodemcuv2\src\pvAlgo.cpp.o
Compiling .pio\build\nodemcuv2\src\rfid.cpp.o
Compiling .pio\build\nodemcuv2\src\shelly.cpp.o
Compiling .pio\build\nodemcuv2\src\webServer.cpp.o
Compiling .pio\build\nodemcuv2\src\webSocket.cpp.o
In file included from src\webServer.cpp:4:
.pio\libdeps\nodemcuv2\AsyncElegantOTA\src/AsyncElegantOTA.h:4:2: error: #error AsyncElegantOTA library is deprecated, Please consider migrating to newer ElegantOTA library which now comes with an async mode. Learn More: https:
    4 | #error AsyncElegantOTA library is deprecated, Please consider migrating to newer ElegantOTA library which now comes with an async mode. Learn More: https://docs.elegantota.pro/async-mode/
      |  ^~~~~
Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\nodemcuv2\lib580\ESP8266WiFi\BearSSLHelpers.cpp.o
Compiling .pio\build\nodemcuv2\lib580\ESP8266WiFi\CertStoreBearSSL.cpp.o
Compiling .pio\build\nodemcuv2\lib580\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pio\build\nodemcuv2\lib580\ESP8266WiFi\ESP8266WiFiAP.cpp.o
In file included from src\webServer.cpp:23:
.pio\libdeps\nodemcuv2\ESP Async WebServer\src/SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
   16 |     SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
      |                                                                                                     ^~~~~~
In file included from C:\Users\cothi\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/CertStoreBearSSL.h:26,
                 from C:\Users\cothi\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:30,
                 from C:\Users\cothi\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiClientSecure.h:23,
                 from C:\Users\cothi\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiServerSecure.h:20,
Compiling .pio\build\nodemcuv2\lib580\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
                 from C:\Users\cothi\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41,
                 from .pio\libdeps\nodemcuv2\AsyncElegantOTA\src/AsyncElegantOTA.h:10,
                 from src\webServer.cpp:4:
C:\Users\cothi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:275:15: note: declared here
  275 | extern fs::FS SPIFFS __attribute__((deprecated("SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems.")));
      |               ^~~~~~
*** [.pio\build\nodemcuv2\src\webServer.cpp.o] Error 1
================================================================================================================================= [FAILED] Took 31.50 seconds =================================================================================================================================
```

cothin, 2024-09-08:
Here is an update regarding the issue:

From whate i can say at this point, it seems to work. At least i was able to build the software. 

After linking AsyncElegantOTA in the lib dependencies. You will have to make sure to comment the error-message in line 4 of the "AsyncElegantOTA_h" out.

The compiling failed again in WebSocketsServer.cpp line 677:

```
.pio\libdeps\nodemcuv2\WebSockets\src\WebSocketsServer.cpp: In member function 'void WebSocketsServer::handleNewClients()':
.pio\libdeps\nodemcuv2\WebSockets\src\WebSocketsServer.cpp:677:82: error: 'class WiFiServer' has no member named 'accept'; did you mean '_accept'?
  677 |     WEBSOCKETS_NETWORK_CLASS * tcpClient = new WEBSOCKETS_NETWORK_CLASS(_server->accept());
      |                                                                                  ^~~~~~
      |                                                                                  _accept
*** [.pio\build\nodemcuv2\lib2e1\WebSockets\WebSocketsServer.cpp.o] Error 1
```

"_accept" isn't working either, but "available()" worked fine for me.

` line 677: WEBSOCKETS_NETWORK_CLASS * tcpClient = new WEBSOCKETS_NETWORK_CLASS(_server->available());
`

I was able to flash the compiled firmware and to setup the wbec service. 
The setup has not been testet at the wallbox and inverter so far. 

I will close this issue, since the build worked for now.

closed 2024-09-08 by cothin

joschro, 2024-09-26:
should be reopened as I ran into the same issue and this needs to be fixed in the code, obviously; would it make sense to split it into two issues, one for AsyncElegantOTA -> ElegantOTA migration and one for "_server->accept()" issue? @steff393