steff393/wbec #146: WBEC reduziert den Strom von 16A auf 6A

Locutus-wolf359, 2024-09-20:
Moin,

da bei uns das das PV Überschußladen nicht mehr immer ausreicht. Wollte ich wieder anfangen mit 11KW laden, weniger Verlust und kürzer.

Wenn ich im EVCC auf Schnelladen stelle habe ich für kurze Zeit 11KW und nach ein paar Sekunden 30-60 reduziert sich die Leistung auf 4,5KW runter. 

Im WBEC wandert zu erst der Schieberegler von 16A auf 6A und etwas später reduziert sich die Leistung auch im EVCC. Das ganze passiert auch, wenn ich Home Assistant mit EVCC komplett abschalte.

Schnellladen haben ich zuletzt im April / Mai benutzt. Da noch ohne Probleme.

Beim WBEC habe ich heute zum testen die 2.5.3 installiert, ohne Änderung.

Danke und Gruß Christian

abuesen, 2024-09-20:
Moin Christian.

Das hatte ich auch, weiß aber nicht mehr genau die Ursache. Es hatte nach meiner dunkeln Erinnerung mit der wbec-eigenen Steuerung und MQTT zu tun. Es ist entstanden, als ich zwischen der Steuerung durch wbec und per MQTT gewechselt habe.

Ich meine, es hatte mit MQTT _maxcurrent_ zu tun. Wenn volle Leistung geladen werden soll schreibt evcc den Wert nicht mehr in das mqtt, damit die Wallbox die eigene, maximale Leistung lädt. Dabei ist bei wbec irgendwie nur 6A raus gekommen, weil es den Wert noch gespeichert hatte. 

Schau mal mit mqtt-explorer, ob das bei Dir auch so ist. _maxcurrent_ kurz da und dann weg.  

Leider weiß ich nicht mehr, wie ich das behoben habe. es war aber ziemlich sicher eine Korrektur am wbec. 
Ich habe auch mal einen Reset gemacht und die Steuerung per MQTT neu eingerichtet. Muss aber nicht bei diesem Fehler gewesen sein.

PS: Wenn der Überschuss nicht ausreicht lade ich trotzdem im Modus PV mit 3P. Dabei ist der Schwellwert in evcc mittels _threshold_ in der Config so eingestellt, dass 1, 2, oder 3kW als ausreichend Überschuss angesehen werden und der Rest kommt aus dem Netz. Also maximale Sonnenausnutzung. Nicht, dass der Akku plötzlich voll ist und es ist noch Sonne da ;) Der Parameter _residual_power_ könnte mit negativem Wert auch reichen. Nicht probiert.

Nachts wird per geplantem Laden mit 11kW der Bedarf noch aufgefüllt, falls die Sonne+Netz nicht gereicht hat. Wenn es kalt ist bekommt die Garage dadurch noch etwas (Ab-)Wärme. Aktuell versuche ich das über die evcc-API mittels HA etwas mehr zu automatisieren, indem ich den minimalen Akkustand in evcc verändere. 

Grüße, Andre.

Locutus-wolf359, 2024-09-21:
Moin Andre,
Danke für deine Antwort.

MQTT nutze ich nicht, bei mir ist die Heidelberg im EVCC als GO-E eingebunden. Spielt MQTT trotzdem im WBEC selbst eine Rolle?

Mich verwundert nur das WBEC auch ohne aktiven EVCC / Home Assistant die Leistung von 16A auf 6A herunter regelt.

Home Assistant ist heruntergefahren, die VM Ware ausgeschaltet. Ladevorgang wird über WBEC Webseite gestartet und auf 16A hoch gestellt. Nach 10-60 Sekunden regelt WBEC die Leistung wieder auf 6A herunter.

Ich versuche auch noch soviel wie möglich an Sonnenenergie ins Auto zu bekommen. Es gibt aber jetzt immer mehr Tage wo das nicht mehr funktioniert. Kommende Woche bei Wolken und Regen z.B. Momentan wird der Sohn wegen gebrochenen Fuß auch per Auto zur Schule / Arzt gefahren.

Gruß Christian

abuesen, 2024-09-23:
Moin moin.

Nein, MQTT benötigt Servereinstellungen und man muss die Regelung mit `cfgPvActive` umschalten. 
Möglicherweise bleiben bei der GoE-API Parameter des letzten Aufrufs aktiv. Schick doch bitte mal Deine JSON (mit Quelltextformatierung) und die Config. In der Config die Username, Passworte und SSID unkenntlich machen.

Grüße, Andre.

Locutus-wolf359, 2024-09-28:
Moin,
hier die JSON:
/{"wbec":{"version":"v2.5.3","bldDate":"Sep 14 2024 11:57:10","timeNow":"20:16:24","enwg14a":0,"enwgErr":0},"box":[{"busId":1,"version":"108","chgStat":4,"currL1":0,"currL2":0,"currL3":0,"pcbTemp":194,"voltL1":241,"voltL2":246,"voltL3":242,"extLock":1,"power":0,"energyP":3490.444,"energyI":6027.983,"energyC":0,"currMax":16,"currMin":6,"logStr":"00.779.2909/02 037700 2215 HD_EnergyControl","wdTmOut":60000,"standby":4,"remLock":1,"currLim":0,"currFs":0,"lmReq":0,"lmLim":0,"resCode":"0","failCnt":0}],"modbus":{"state":{"lastTm":630136748,"millis":630141195}},"rfid":{"enabled":false,"release":false,"lastId":""},"pv":{"mode":0,"watt":0,"wbId":0},"wifi":{"mac":"0C:8B:95:A6:14:44","rssi":0,"signal":100,"channel":0}}

```javascript
{"cfgApSsid":"wbec","cfgApPass":"wbec1234","cfgCntWb":1,"cfgMbCycleTime":10,"cfgMbDelay":100,"cfgMbTimeout":60000,"cfgStandby":4,"cfgFailsafeCurrent":0,"cfgMqttIp":"","cfgMqttPort":1883,"cfgMqttUser":"","cfgMqttPass":"","cfgMqttWattTopic":"wbec/pv/setWatt","cfgMqttWattJson":"","cfgNtpServer":"europe.pool.ntp.org","cfgFoxUser":"","cfgFoxPass":"","cfgFoxDevId":"","cfgPvActive":0,"cfgPvCycleTime":30,"cfgPvLimStart":61,"cfgPvLimStop":50,"cfgPvPhFactor":69,"cfgPvOffset":0,"cfgPvCalcMode":0,"cfgPvInvert":0,"cfgPvInvertBatt":0,"cfgPvMinTime":0,"cfgPvOffCurrent":255,"cfgPvHttpIp":"","cfgPvHttpPath":"/","cfgPvHttpJson":"","cfgPvHttpPort":80,"cfgTotalCurrMax":0,"cfgLmChargeState":4,"cfgRestoreLastReq":0,"cfgHwVersion":15,"cfgWifiSleepMode":0,"cfgLoopDelay":255,"cfgKnockOutTimer":0,"cfgShellyIp":"","cfgInverterIp":"","cfgInverterType":0,"cfgInverterPort":0,"cfgInverterAddr":0,"cfgInvSmartAddr":0,"cfgInvRegPowerInv":0,"cfgInvRegPowerInvS":0,"cfgInvRegPowerMet":0,"cfgInvRegPowerMetS":0,"cfgInvRegToGrid":0,"cfgInvRegFromGrid":0,"cfgInvRegInputGrid":0,"cfgInvRegBattery":0,"cfgBootlogSize":2000,"cfgBtnDebounce":0,"cfgWifiConnectTimeout":10,"cfgResetOnTimeout":0,"cfgEnergyOffset":0,"cfgDisplayAutoOff":2,"cfgWifiAutoReconnect":1,"cfgWifiScanMethod":0,"cfgLedIp":1,"cfgWifiOff":1,"cfgChargeLog":0}
```

abuesen, 2024-09-30:
Guten Morgen!

Da sehe ich erstmal nichts Auffälliges.
Ich würde als nächstes Werkseinstellungen nutzen und dann schauen, wenn @steff393 da nicht noch was erkennen kann.

BTW: Im Auto hast Du bestimmt geschaut, ob da irgendwelche Begrenzungen eingestellt sind.

Grüße.

abuesen, 2024-10-01:
Abend!

> Wie macht man einen Werksreset? In der Anleitung hatte ich dazu nichts gefunden.

Stimmt, gibt es nicht in der Oberfläche. Dann hab ich das wohl auch nicht gemacht. Oder irgendwo eine Standard-Config geholt... Ist zu lange her in meinem Alter :)
 
> EDIT: Könnte ich mir erstmal behelfen in dem ich den currMin: 6 in der json einfach auf 16 stelle?

Ausprobieren.  Ansonsten weiß ich dann auch nicht weiter.

LG.

abuesen, 2024-10-01:
Das gesamte System inklusive Wallbox ist bestimmt einmal neu gestartet. Alles aus, erst Wallbox dann wbec einschalten. Nur zur Sicherheit...

Locutus-wolf359, 2024-10-01:
> Das gesamte System inklusive Wallbox ist bestimmt einmal neu gestartet. Alles aus, erst Wallbox dann wbec einschalten. Nur zur Sicherheit...

Bisher habe ich WBEC nur neugestartet. Jetzt habe ich Wallbox und WBEC ausgeschaltet, gewartet und neu gestartet. Leider keine Änderung.

Moin,
> evcc und wbec sind nicht über MQTT verbunden, sondern vermutlich über die go-e-Charger-Emulation, oder?

Richtig, MQTT habe ich nicht am laufen. Im EVCC habe ich die Heidelberg als Go-E angegeben.

> Zeitsteuerung oder ähnliches ist nicht aktiv, oder?

Nein

> Bei Auslieferung schreib ich in die cfg.json immer nur {"cfgApSsid":"wbec","cfgApPass":"wbec1234"} rein. Alles was da nicht extra aufgeführt ist bleibt dann einfach auf Default-Werten.

Bei mir steht das drin (WBEC-Lan):
{"cfgApPass":"wbec1234","cfgCntWb":1,"cfgWifiOff":1}

Was ich nicht so ganz verstehe, wenn EVCC und HomeAssistant komplett Offline und Stromlos sind. Schaltet WBEC trotzdem den Strom nach ein paar Sekunden wieder zurück von 16A auf 6A. Gesartet und auf 16A über das WBEC Webinterface gestellt.

EVCC fängt auch bei Auswahl "schnell" immer mit 6A an und erhöht die Leistung nicht. Das war sonst anders.

Trage ich in EVCC min 16A und Max auf 16A ein, dann startet der Ladevorgang mit 16A, hört dann aber nach 10-60 Sekunden auf. Da im WBEC der Schieberegler wieder auf 6A zurück geht und EVCC dann einen nicht erwarteten Wert meldet und den Ladevorgang stoppt.

Ich werde morgen mit dem Auto noch mal an einer öffentlichen 11KW Ladesäule halten und beobachten. Nicht das es am Fahrzeug liegt.

Locutus-wolf359, 2024-10-07:
Am Auto liegt es nicht.

Habe heute einen Firmenwagen mit nach Hause genommen. Dieser lädt ebenso 30-60 Sekunden mit 11kw und schaltet dann auf 4,4kw herunter.

In der Firma wird durchgängig mit 11kw geladen.

Locutus-wolf359, 2024-10-13:
@steff393 
Kann ich WBEC irgendwie "zwingen" nur noch mit 16A zu laden?

steff393, 2024-10-13:
Wenn du wbec
- per Schieberegler auf 16A oder
- per http://wbec.local/json?currLim=160
einstellst und keine sonstige Steuerung (z.B. PV-Laden, MQTT, Zeitsteuerung, ...) "dazwischenfunkt", dann stellt wbec die 16A als "max. Ladestrom" ein.

Das Auto entscheidet dann, wieviel Strom es dann von der Wallbox zieht.

abuesen, 2024-11-13:
@Locutus-wolf359 in einem gleich gelagerten Fall war die Wallbox noch auf 6A eingestellt und hat dadurch den wbec  korrekt zurück gesetzt.

Schalter S1 muss auf die Pos. 5 .

closed 2025-01-05 by steff393