steff393/wbec #151: Ladestrom lässt sich nicht einstellen; Laden erlaubt: nein

pe-cloud, 2024-11-01:
Hallo, ich benötige etwas Unterstützung bei meinem wbec (ESP8266)
Aktuelle Firmware ist installiert. 

Ich nutze die Wallbox nie; habe diese günstig erhalten und installieren können und wollte mich für die Elektromobilität vorbereiten. Dabei habe ich bei der Installation nach einer Möglichkeit gesucht, wie ich Verbrauchsmessung und Steuerung digital vornehmen kann. Da bin ich auf wbec gestoßen. PV Überschussladen spielt bei mir keine Rolle und ist deaktiviert. 

Aktuell habe ich Gäste, die mit E-Auto da sind und die Möglichkeit zum Laden nutzen möchten. Leider kann ich das Laden über den Schieberegler nicht aktivieren. Dieser springt immer wieder auf den Wert null zurück. Gleichzeitig steht bei Laden erlaubt "nein". Warum dies so ist, ist mir ein Rätsel. 
Gleichzeitig habe ich auch bemerkt, dass mir die Weboberfläche teilweise unterschiedlich angezeigt wird. Ich sende hier einen Screenshot mit, der die aktuell dominierende Ansicht darstellt. 
Kurzum: Ich will einfach nur das Laden des Fahrzeugs starten und steuern können. Danke für die Hilfe! 
[image]

pe-cloud, 2024-11-01:
Nachdem ich versucht habe die mit der letzten Firmware verbreiteten Dateien noch mal per Upload zu aktualisieren, geht aktuell gar nichts mehr! Ich komme nicht mehr auf die Oberfläche. Wie flashe ich das Ganze, um es wieder lauffähig zu bekommen?

steff393, 2024-11-01:
Hallo @pe-cloud 
ohne deine Einstellungen (cfg.json) ist das schwierig zu beurteilen. Ich empfehle den Kauf der [Fertigprodukte](https://steff393.github.io/wbec-site/), statt selbst zu basteln.

pe-cloud, 2024-11-01:
Die aktuellen Einstellungen der cfg.json: 
{"cfgApSsid":"wbec","cfgApPass":"wbec1234","cfgCntWb":1,"cfgMqttIp":"","cfgMqttLp":[],"cfgTotalCurrMax":110}

Was heißt hier Fertigprodukte? Zur Zeit meiner "Bestellung" gab es nur dieses Modell bzw. Möglichkeit. Ich habe hier an der Hardware nichts selbst gebastelt.

steff393, 2024-11-01:
Ach so, ok verstanden, dachte es handelt sich um ein selbstgebautes wbec.
Bitte entferne den Parameter cfgTotalCurrMax:
`{"cfgApSsid":"wbec","cfgApPass":"wbec1234","cfgCntWb":1,"cfgMqttIp":"","cfgMqttLp":[]}`

pe-cloud, 2024-11-01:
Danke! Das hat schon mal geholfen den Ladevorgang zu starten. Allerdings beschränkt sich der Schieberegler auf einen Ladestrom von 6A. Ziehe ich diesen auf einen höheren Wert, rutscht dieser autom. auf 6A zurück. Wo kann ich das einstellen bzw. ändern?

steff393, 2024-11-02:
Probier mal bitte http://wbec.local/json?currLim=160   (mit großem "L") und schicke mir die Antwort.
wbec.local ggf. durch die IP ersetzen.

pe-cloud, 2024-11-02:
Der Wert für currLim bleibt auch beim Aufruf der Adresse auf dem Wert 60: 

`{
  "wbec": {
    "version": "v0.5.3",
    "bldDate": "Sep 14 2024 11:19:17",
    "timeNow": "16:20:12"
  },
  "box": [
    {
      "busId": 1,
      "version": "108",
      "chgStat": 2,
      "currL1": 0,
      "currL2": 0,
      "currL3": 0,
      "pcbTemp": 233,
      "voltL1": 234,
      "voltL2": 233,
      "voltL3": 233,
      "extLock": 1,
      "power": 0,
      "energyP": 25.671,
      "energyI": 25.672,
      "energyC": 25.671,
      "currMax": 6,
      "currMin": 6,
      "logStr": "00.779.2909/01 015453 2136 HD_EnergyControl",
      "wdTmOut": 60000,
      "standby": 4,
      "remLock": 1,
      "currLim": 60,
      "currFs": 0,
      "lmReq": 160,
      "lmLim": 255,
      "resCode": "0",
      "failCnt": 0
    }
  ],
  "modbus": {
    "state": {
      "lastTm": 90042732,
      "millis": 90049918
    }
  },
  "rfid": {
    "enabled": false,
    "release": false,
    "lastId": ""
  },
  "pv": {
    "mode": 0,
    "watt": 0,
    "wbId": 0
  },
  "wifi": {
    "mac": "94:B9:7E:14:54:B1",
    "rssi": -70,
    "signal": 60,
    "channel": 1
  }
}`

steff393, 2024-11-02:
"currMax": 6, "currMin": 6
In deiner Wallbox ist vermutlich der Schalter für den maximalen Strom noch falsch eingestellt.
Der Schalter S1 sollte in der Regel auf >= 5 stehen, d.h. 16A (ggf. je nach Fahrzeug anpassen)