steff393/wbec #153: pvMode über MQTT

carlos22, 2024-12-14:
Hallo,

ich baue mir gerade ein Steuerungsdisplay mit dem CYD und ESPHome (beispiele hier: https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/ESPHome - ich will aber LVL verwenden weil das auch slider und so kann). 

Wäre es möglich auch den pvMode (1 = aus, 2 = PV, 3 =PV+Min) via MQTT abruf und einstellbar zu machen? Wäre schön das auch direkt im HA sehen und ändern zu können und die integration via ESPHome ist halt sehr einfach zu bauen dann. Alle anderen wichtigen Daten habe ich da ja bereits da.

Kann auch gerne die yaml config dann mal posten hier für interessierte...

blob810, 2024-12-14:
Ist bereits integriert. Du musst nur folgendes in das Autodiscovery-Script hinzufügen: 

` - data:
          topic: homeassistant/sensor/wbec/pvMode/config
          retain: true
          payload: |
            {
              "state_topic": "wbec/pv/mode",
              "name": "PV Laden",
              "unique_id": "pvMode",
              "object_id": "wbec_pvMode",
              "icon": "mdi:solar-power-variant",
              "availability_topic":"wbec/connection",
              "payload_available":"online",
              "payload_not_available":"offline",
              "device": {
                  "identifiers": ["wbec"],
                  "name": "Heidelber Energy Control via wbec",
                  "model": "HD_EnergyControl",
                  "manufacturer": "HEIDELBERG via wbec",
                  "sw_version": "1.0.0"
              }
            }
        action: mqtt.publish`

Dadurch wird der Wert zumindest übermittelt. Setzen kannst du den Wert von HA per Rest Command. Ich habe mir über ein paar Helfer selbst etwas gebaut. 

[image]

carlos22, 2024-12-24:
Ah ok cool danke :) Bin noch nicht so firm in HA kannst du kurz skizieren wie du das gemacht hast mit den Helpern?

closed 2025-05-18 by steff393

carlos22, 2025-10-26:
@steff393 Mode setzten über MQTT ist also nicht geplant richtig? Wäre aus meiner Sicht schon sinnvoll das man nicht mit zwei APIs hantieren muss...