steff393/wbec #154: Ladestrom immer auf 0A nach Stromausfall

knebel99, 2025-01-02:
Hallo,

ist mir heute wieder passiert, ich musste den Strom abschalten im Haus, danach hat die WB nicht mehr geladen.
Ich musste den Strom manuell wieder auf 16A (max.) stellen da er auf 0A war.

Habs danach nochmal probiert, anscheinend setzt das Modul den Ladestrom immer auf 0A bei nem Neustart.
Hab nichts besonderes eingestellt, keine Überschussladen oder MQTT etc. (noch nicht ;-)

Kann man das irgendwo konfigurieren oder ist das ein Bug?

steff393, 2025-01-03:
Hi, ist ein Feature ;-)
Siehe hier die Anleitung der Heidelberg:
[image]

knebel99, 2025-01-03:
Oh OK vielen Ank für die Info. Also nix was man in wbec umbiegen könnte?

steff393, 2025-01-03:
Technisch wäre es schon möglich. Aber das war eine der allerersten Design-Entscheidungen: Dass der gültige Stromwert immer der ist, der aktuell in der Wallbox steht. Das zu ändern würde einiges umwerfen und ist daher aktuell nicht im Fokus.

closed 2025-01-03 by steff393

knebel99, 2025-01-03:
Gut trotzdem vielen Dank für die Erklärung :-)