steff393/wbec #155: WBEC Pro Kommunikationspropblem mit EVCC (via MQTT) -->  geht in Offline / Standby

Eversman13, 2025-01-27:
Hallo zusammen,

ich habe aktuell ein paar Probleme mit EVCC und wbec; es werden keine Daten von EVCC nach wbec übertragen, obwohl alles gemäß Anleitung unter https://github.com/steff393/wbec/wiki/Configuration-Example%3A-EVCC-via-MQTT konfiguriert ist. Ich habe alles x-fach kontrolliert, im MQTT-Explorer kann ich alle Topcis und Werte sehen (auch, dass sie sich regelmäßig aktualisieren) - nur bei webec kommt nichts an. 

Meine Vermutung ist nun, dass wbec keine Steuerung aus EVCC entgegen nimmt, weil wbec sich nach kurzer Zeit in den Status "offline / standby" setzt (siehe screenshot). Auch über die GUI ist dann keine Steuerung mehr möglich.

[image]

Ist das ein normales / gewünschtes Verhalten? Habe ich da was übersehen? 

P.S.: falls jemand wbec und EVCC erfolgreich via MQTT verbunden hat, bitte gerne melden...

abuesen, 2025-01-27:
Hi.

Für die Steuerung muss ein Auto angeschlossen sein. Im Screenshot ist keines verbunden.

In wbec muss die PV Steuerung deaktiviert sein, sonst rechnet die Software selbst aus und steuert den Ladestrom.

Irgendwann zum Start hatte ich auch mal das Problem, dass maxCurrent nicht geschrieben wurde und somit auch der Ladevorgang ausblieb. Ansonsten läuft das gut.

Eversman13, 2025-01-28:
Das wäre natürlich ein einfacher (und von mir sehr dummer) Grund.. ich teste das beim nächsten Ladevorgang. Danke!

Eversman13, 2025-01-28:
Ich hatte tatsächlich noch einen Fehler in der config.. ich habe nochmal alles kopmplett zurückgesetzt und bin mit einer "leeren" config gestartet uns siehe da: alles funktioniert!

closed 2025-01-28 by Eversman13