steff393/wbec #156: Feature Request: cfgPvPhFactor berechnen

hmei67, 2025-02-10:
Nach meinem Verständnis ist es wichtig, cfgPvPhFactor=23 bei einphasigem Anschluss bzw. = 69 bei dreiphasigem Anschluss manuell zu setzen, damit der interne Algo die passende Stromstärke berechnen kann. 

Wbec kennt anscheinend nicht nur die Ströme pro Phase, sondern auch die zugehörigen Spannungen. Könnte wbec dann nicht diesen Faktor daraus selbständig berechnen? 

Beispiel: meine EC ist einphasig angeschlossen und nur die L1 hat 230V, die anderen beiden zwischen 8V und 45V (warum auch immer) => cfgPvPhFactor=23. Wenn ich per Umschalter die EC dreiphasig mache (per Umschalter im Zählerschrank wenn kein Auto angesteckt ist), dann cfgPvPhFactor=69. 

Dann bräuchte ich das nicht jedesmal händisch zu machen...

Gruß
  Hermann

steff393, 2025-02-10:
Es gibt für den Parameter cfgPvPhFactor tatsächlich 2 Spezialwerte (bislang undokumentiert und auch kein offizielles Feature):
cfgPvPhFactor = 0: 
- Wenn die Spannung an L1 > 200V und L2, L3 < 15V, dann 1-phasig, sonst 3-phasig
- Erkennt also wie die Wallbox angeschlossen ist

cfgPvPhFactor = 1: 
- Wertet den Strom an L1, L2 und L3 (in dieser Reihenfolge)  aus. Wenn Strom > 0, dann wird die Phase als aktiv betrachtet und somit cfgPvPhFactor intern quasi als 23, 46 oder 69 betrachtet.
- Erkennt also, welches Fahrzeug angeschlossen ist (allerdings immer erst nach dem Ladebeginn, braucht also kurz zum Einregeln)

Der Wert 15V ist leider hartkodiert und nicht änderbar. Bei waren das stets einstellige Spannungswerte, wenn die Phase nicht angeschlossen ist.

VedMod, 2025-02-11:
Also zumindest mit dem veröffentlichen code geht das aber nicht gut.

Hier wird im Fall 1 dann aus Leistung 1:1 Stromstärke gemacht :D ;)
https://github.com/steff393/wbec/blob/3d573f3ebce7a217fa1f515df76c9337bf82b58c/src/pvAlgo.cpp#L47C3-L47C6

steff393, 2025-02-11:
Hast recht. Das betrifft wbecPro, wbecLan & wbecPremium.

closed 2025-02-11 by steff393