steff393/wbec #157: Keine Verbindung nach Stromausfall

abuesen, 2025-02-20:
Hi steff.

Bei uns gab es wieder einen Stromausfall für rund 5 Minuten.

Nach Wiedereinschaltung kam die Kommunikation zwischen wbec (LAN) und HDEC nicht zustande. Die WB blinkt weiß-blau und zeigt den fehlenden Modbus-Master an. Neustart der WB war ohne Änderung des Zustandes. Neustart des wbec hat die Kommunikation wiederhergestellt.

Zusätzlich war der wbec auch nicht per IP erreichbar. Klar, der Router braucht einige Minuten zum Starten des DHCP-Servers.

Das gesamte Verhalten ist mir bei anderen Stromausfällen bisher nicht aufgefallen und wir hatten kürzlich einige, als irgendwelche Überlandleitungen vereist waren. Ich kann mich dunkel erinnern, dass ganz zu Anfang einmal das Geblinke der WB da war und ich auch den wbec neu gestartet habe. Mindestens 18 Monate her.

Hast Du eine Idee dazu? Wie kann man das Wiedereinschalte stabil bekommen?

Grüße aus der Eifel
Andre.

anselra, 2025-02-21:
Hi Andre,

schau mal hier das könnte dir helfen 
https://github.com/steff393/wbec/issues/139#issuecomment-2187242915

Gruß Ralf

abuesen, 2025-02-21:
Moin Ralf,

danke, hab WLAN ausgeschaltet und wbec ist per Kabel angebunden. Beim Kabel kann ich notfalls den Switch-Port nach einer Internetverbindung deaktivieren/aktivieren. Müsste bei meinem Switch zumindest gehen. Noch nicht nach geschaut. Eher doof ist die Modbus-Seite. Da kam ebenfalls keine Kommunikation zu Stande.

Ich denke zB an `cfgKnockOutTimer` weiß aber nicht, was der bei aktiven Ladevorgängen für Nebeneffekte hat. 
Ein Art Watchdog oder `cfgKnockOutTimer` in Verbindung mit einer `netProbeIP` wäre edel, gibt es aber mW nicht.

Gruß Andre.

steff393, 2025-05-18:
cfgKnockOutTimer hat keinen Nachteil bei aktiven Ladevorgängen. wbec ist nach einem Reset innerhalb weniger Sekunden wieder im Normalzustand -> das merkt die Wallbox gar nicht.

timpalm, 2025-10-18:
Wie es der Zufall so will hatten wir auch einen Stromausfall. Danach hat sich der wbec nicht mit dem WLAN connected. Musste später die Sicherung raus nehmen und neu starten und - dann klappt alles. Kann es sein, wenn das WLAN nicht zeitig verfügbar ist, dass er es nicht mehr findet bzw durch einen timeout nicht mehr versucht

steff393, 2025-10-24:
Ja, das ist leider eine Schwäche. Abhilfe wäre der Parameter
uint16_t cfgWifiConnectTimeout;       // Timeout in seconds to connect to Wifi before change to AP-Mode
Defaultwert sind 10s. Wenn der Router länger braucht, dann ggf. diese Zeit erhöhen.

closed 2025-10-24 by steff393