steff393/wbec #165: curl bricht bei Update ab: 404 Not Found

abuesen, 2025-08-16:
Windows 11 24H2

curl 8.13.0 (Windows) libcurl/8.13.0 Schannel zlib/1.3.1 WinIDN
Release-Date: 2025-04-02

wbec version "v2.5.2"

curl -v --data-binary @"wbecLan_firmware_2_5_4.bin" http://arduino:password@192.168.123.169/sketch
*   Trying 192.168.123.169:80...
* Connected to 192.168.123.169 (192.168.123.169) port 80
* using HTTP/1.x
* Server auth using Basic with user 'arduino'
> POST /sketch HTTP/1.1
> Host: 192.168.123.169
> Authorization: Basic YXJkdWlubzpwYXNzd29yZA==
> User-Agent: curl/8.13.0
> Accept: */*
> Content-Length: 1270464
> Content-Type: application/x-www-form-urlencoded
> Expect: 100-continue
>
* Done waiting for 100-continue
* HTTP 1.0, assume close after body
< HTTP/1.0 404 Not Found
< Content-Type: text/plain
< Content-Length: 12
< Connection: close
<
FileNotFound* we are done reading and this is set to close, stop send
* abort upload after having sent 131072 bytes
* shutting down connection #105

abuesen, 2025-08-16:
Dachte gerade, vielleicht Updates der Reihe nach. Bei 2.5.3 kommt der gleiche Fehler.

abuesen, 2025-08-17:
Hab mich heute nochmal etwas damit beschäftigt.
Hier muss entweder der Pfad falsch sein, der steht aber so in der Anleitung, oder der content-type wird falsch geraten.

Sollte es nicht eher "Content-Type: application/octet-stream" sein?

VedMod, 2025-08-17:
Mit was führst du curl aus? "Eingabeaufforderung" oder PowerShell?

Ich meine ich hätte bei einem Update auch schon mal ähnliches phenomän gehabt und dann die gute alte "Eingabeaufforderung" genommen und damit ging alles. (Oder es war anders rum :D ;) )

abuesen, 2025-08-17:
Mit cmd. Powershell-curl mag nicht.

Wer lesen kann... Port vergessen.
Mit dem richtigen Port geht es.

closed 2025-08-17 by abuesen